SmallChi/JT808 #13: JT808MessagePackReader decode中为什么还要产生一个Span<byte> allocateBuffer呢

491134648, 2020-04-03:
JT808MessagePackReader decode中为什么还要产生一个Span<byte> allocateBuffer呢，直接读SrcBuffer不是更能减少一次内存分配吗，
`
   /// <summary>
        /// 在解码的时候把校验和也计算出来，避免在循环一次进行校验
        /// </summary>
        /// <returns></returns>
        public void Decode(Span<byte> allocateBuffer)
        {
            int i = 0;
            int offset = 0;
            int len = SrcBuffer.Length;
            _realCheckXorCode = 0;
            allocateBuffer[offset++] = SrcBuffer[0];
            // 取出校验码看是否需要转义
            ReadOnlySpan<byte> checkCodeBufferSpan = SrcBuffer.Slice(len - 3,2);
            int checkCodeLen = 1;
            if (checkCodeBufferSpan.SequenceEqual(decode7d01))
            {
                _realCheckXorCode = 0x7d;
                checkCodeLen += 1;
            }
            else if (checkCodeBufferSpan.SequenceEqual(decode7d02))
            {
                _realCheckXorCode = 0x7e;
                checkCodeLen += 1;
            }
            else
            {
                _realCheckXorCode = checkCodeBufferSpan[1];
            }
            len = len - checkCodeLen - 1 - 1;
            ReadOnlySpan<byte> tmpBufferSpan = SrcBuffer.Slice(1, len);
            while (i < len)
            {
                if (tmpBufferSpan[i] == 0x7d)
                {
                    if (len > i + 1)
                    {
                        if (tmpBufferSpan[i + 1] == 0x01)
                        {
                            allocateBuffer[offset++] = 0x7d;
                            _calculateCheckXorCode = (byte)(_calculateCheckXorCode ^ 0x7d);
                            i++;
                        }
                        else if (tmpBufferSpan[i + 1] == 0x02)
                        {
                            allocateBuffer[offset++] = 0x7e;
                            _calculateCheckXorCode = (byte)(_calculateCheckXorCode ^ 0x7e);
                            i++;
                        }
                        else
                        {
                            allocateBuffer[offset++] = tmpBufferSpan[i];
                            _calculateCheckXorCode = (byte)(_calculateCheckXorCode ^ tmpBufferSpan[i]);
                        }
                    }
                }
                else
                {
                    allocateBuffer[offset++] = tmpBufferSpan[i];
                    _calculateCheckXorCode = (byte)(_calculateCheckXorCode ^ tmpBufferSpan[i]);
                }
                i++;
            }
            allocateBuffer[offset++] = _realCheckXorCode;
            allocateBuffer[offset++] = SrcBuffer[SrcBuffer.Length- 1];
            _checkXorCodeVali = (_calculateCheckXorCode == _realCheckXorCode);
            Reader = allocateBuffer.Slice(0, offset);
            _decoded = true;
        }
`
改成以下不是更好吗
`
  public void Decode()
        {
            int i = 0;
            int len = Reader.Length;
            // 取出校验码
            ReadOnlySpan<byte> checkCodeBufferSpan = Reader.Slice(len - 3, 2);
            _realCheckXorCode =BinaryPrimitives.ReadUInt16LittleEndian(checkCodeBufferSpan);
            int checkCodeLen = 2;
            len = len - checkCodeLen;
            ReadOnlySpan<byte> tmpBufferSpan = Reader.Slice(0, len);
            _calculateCheckXorCode = CRC.CalcBufferDataCrc161(tmpBufferSpan);
            _checkXorCodeVali = (_calculateCheckXorCode == _realCheckXorCode);
            //Reader = allocateBuffer.Slice(0, offset);
            _decoded = true;
        }
`

SmallChi, 2020-04-03:
你先认真看下协议，校验码也有可能有转义字符，不是凭空写的。╮（╯＿╰）╭

closed 2020-04-03 by SmallChi

491134648, 2020-04-06:
好吧，我拿来用的，不是JT808协议，我去掉应该没关系吧

SmallChi, 2020-04-06:
嗯，根据自己协议去处理久好了。