SmallChi/JT808 #18: 关于某些消息的位操作的疑问

Luke-Tang, 2020-11-25:
用2011的车辆控制消息做例子:
7e85000001013512344321303901fc7e
JTTOOL解析的部分结果:
"数据体对象": {
    "车辆控制": "01",
    "[01]控制标志": 1,
    "控制标志对象[00000001]": {
      "[bit1~bit7]保留": "0000001",
      "[bit0]": "车门解锁"
    }
  },

按照协议文档，控制标志的bit0的值： 0为车门解锁 ； 1为车门加锁
上面消息解析后的控制标志为: [00000001]
按我的理解，一个字节位的对应关系应该如下:
 bit:    7 6 5 4 3 2 1 0
         [0 0 0 0 0 0 0 1]
这样bit0的值为1，应为车门加锁。

但项目中协议分析部分的代码并不是我理解的样子。而是如下:
bit:   0 1 2 3 4 5 6 7
       [0 0 0 0 0 0 0 1]
这样bit0的值为0，为车门解锁。和JTTool的解析相符。

还有几条涉及到位操作的协议也是如此。 请问是我的理解有误，还是这是项目的BUG

SmallChi, 2020-11-25:
不支持2011版本

Luke-Tang, 2020-11-25:
2011版本和2013版本此协议是一样的。关键点在于位的顺序。
bit: 7 6 5 4 3 2 1 0
[0 0 0 0 0 0 0 1]  
这是正确的还是下面是正确的
bit: 0 1 2 3 4 5 6 7
[0 0 0 0 0 0 0 1]

第一种和包的消息体属性是一致的

SmallChi, 2020-11-25:
上面是对的

Luke-Tang, 2020-11-25:
那对应的消息的Analyze方法中，对位的解释就有问题了。
JT808_0x8500.cs

  ReadOnlySpan<char> controlFlagBits = Convert.ToString(value.ControlFlag, 2).PadLeft(8, '0').AsSpan();
  writer.WriteStartObject($"控制标志对象[{controlFlagBits.ToString()}]");
  writer.WriteString("[bit1~bit7]保留", controlFlagBits.Slice(1, 7).ToString());
  writer.WriteString("[bit0]", controlFlagBits[0]=='0'? "车门解锁" : "车门加锁");

这段代码应该是把字节转成String了，  字节0x01转成了"000000001"
那么controlFlagBits[0]其实取的是最高位了

SmallChi, 2020-11-25:
嗯  有问题。

Luke-Tang, 2020-11-25:
有好几条消息都有这个问题。
非常感谢提供JTTool，测试协议非常方便

SmallChi, 2020-11-25:
哪些有问题可以提出来，改下就行了。

Luke-Tang, 2020-11-26:
我所知道的：
0x8203
0x8300
0x8500

和位操作相关的有可能都存在这个问题

SmallChi, 2020-11-30:
已修复

closed 2020-11-30 by SmallChi