SmallChi/JT808 #26: JT808Serializer().HeaderDeserialize()解包头异常

DonPangPang, 2021-11-06:
`JT808Serializer().HeaderDeserialize()`在解析自己组装的2019包时，Header解析错误，采用`JT808Serializer().Deserialize()`解析正常

SmallChi, 2021-11-06:
请大兄弟提供对应的原始数据，方便排查。

DonPangPang, 2021-11-06:
查询终端参数：
`7E81040000010000000001234567891000011C7E`
```
//组包代码
JT808Package jT808Package =
                    JT808MsgId.查询终端参数.Create(pNo,
                        new JT808_0x8104());
                var buffer = new JT808Serializer().Serialize(jT808Package, JT808Version.JTT2019);
                var str = buffer.ToHexString();//7E81040000010000000001341990550700016B7E
```
采用`JT808Serializer().HeaderDeserialize()`进行解析，手机号解析为`10000000001`[错误]
采用`JT808Serializer().Deserialize()`进行解析，手机号为`12345678910`[正常]

其它：
在JttTools中，该包同样会被解析为2013版本，且手机号为`10000000001`[错误]

DonPangPang, 2021-11-06:
> 请大兄弟提供对应的原始数据，方便排查。

好的老哥~

DonPangPang, 2021-11-06:
> 请大兄弟提供对应的原始数据，方便排查。

查询终端参数：
`7E81040000010000000001234567891000011C7E`
```
//组包代码
JT808Package jT808Package =
                    JT808MsgId.查询终端参数.Create(pNo,
                        new JT808_0x8104());
                var buffer = new JT808Serializer().Serialize(jT808Package, JT808Version.JTT2019);
                var str = buffer.ToHexString();//7E81040000010000000001341990550700016B7E
```
采用`JT808Serializer().HeaderDeserialize(buffer, JT808Version.JTT2019)`进行解析，手机号解析为`10000000001`[错误]
采用`JT808Serializer().Deserialize(buffer, JT808Version.JTT2019)`进行解析，手机号为`12345678910`[正常]

其它：
在JttTools中，该包同样会被解析为2013版本，且手机号为`10000000001`[错误]

DonPangPang, 2021-11-06:
> 请大兄弟提供对应的原始数据，方便排查。

顺便来一张图，做了下对比：
[image]

SmallChi, 2021-11-06:
JT808MsgId.查询终端参数.Create不是用了2019的版本进行组包的，你应该用JT808MsgId.查询终端参数.Create_2019这样才是2019版本。

DonPangPang, 2021-11-06:
> JT808MsgId.查询终端参数.Create不是用了2019的版本进行组包的，你应该用JT808MsgId.查询终端参数.Create_2019这样才是2019版本。

在`JT808Serializer().Serialize()`的时候指定了版本，而且使用`Deserialize()`解析为2019时解析正常啊。。这里面有什么说法吗？

SmallChi, 2021-11-06:
有可能真是bug喔  🌚

DonPangPang, 2021-11-06:
> 有可能真是bug喔 🌚

我觉得可能是，而且在老哥你的JttTools里，一些2019的包也会识别成2013版本，类似于0x0200这些定位的包解析是正常的，但是一些类似于0x8104的这些命令包，就会识别成2013版本。

此外0x0702（驾驶员身份信息采集上报），这个包解析有问题，我暂时还需要研究一下，是不是由于海康设备对这个协议进行了改动。

closed 2021-11-06 by waterliu99