SmallChi/JT808 #33: JT808MessagePackWriter.WriteChar

codlin, 2023-11-20:
```C#
public void WriteChar(in char value)
{
    var span = writer.Free;
    span[0] = (byte)value;
    writer.Advance(1);
}
```
char 占两个字节吧，对于非 ASCII 字符，如汉字，是不是会有问题？

SmallChi, 2023-11-20:
汉字走的是gb2312编码。

closed 2023-11-20 by SmallChi

yedajiang44, 2023-11-21:
@codlin char占用几个字节跟字符编码有关系，因此你说的2个字节是不对的，但是协议规定用GBK和ASCII（其实GBK已经包含了ASCII），因此无论如何都是1个字节（具体原理可以研究下GBK编码），所以你大可不必担心有什么问题

可尝试如下代码：

```c#
System.Text.Encoding.ASCII.GetByteCount(new char[] { 'a' }); //1
System.Text.Encoding.GetEncoding("GBK").GetByteCount(new char[] { 'a' }); //1
System.Text.Encoding.Unicode.GetByteCount(new char[] { 'a' }); //2
```

codlin, 2023-11-21:
> @codlin char占用几个字节跟字符编码有关系，因此你说的2个字节是不对的，但是协议规定用GBK和ASCII（其实GBK已经包含了ASCII），因此无论如何都是1个字节（具体原理可以研究下GBK编码），所以你大可不必担心有什么问题
> 
> 可尝试如下代码：
> 
> ```cs
> System.Text.Encoding.ASCII.GetByteCount(new char[] { 'a' }); //1
> System.Text.Encoding.GetEncoding("GBK").GetByteCount(new char[] { 'a' }); //1
> System.Text.Encoding.Unicode.GetByteCount(new char[] { 'a' }); //2
> ```

好的，我后面研究一下 GBK 编码。多谢了~