SmallChi/JT808 #34: Support for docker environment

rockenbach, 2023-12-06:
Hey, congratulations on this really nice project!

Currently, as a studying practice, I'm trying to develop a python parser for JT808. I do not have experience with dotnet platform, but I would like to use this project as a result comparison.

VSCode had added a dockerfile and docker compose for the project, but I was only able to execute `docker-compose up` after change the file `src/JT808.Protocol/JT808.Protocol.csproj` to add the following inside Project > PropertyGroup:

```
<TargetFramework>net8.0</TargetFramework>
<RuntimeIdentifier>linux-x64</RuntimeIdentifier>
<GenerateRuntimeConfigurationFiles>true</GenerateRuntimeConfigurationFiles>
```
Still, I'm getting the following error when I execute `docker compose up`:

`Unhandled exception. System.MissingMethodException: Entry point not found in assembly 'JT808.Protocol, Version=2.6.4.0, Culture=neutral, PublicKeyToken=null'.`

Can someone help me with some tips on how to use this project?

<details><summary>generated src/JT808.Protocol/Dockerfile</summary>
<p>

```
FROM mcr.microsoft.com/dotnet/runtime:8.0 AS base
WORKDIR /app

USER app
FROM --platform=$BUILDPLATFORM mcr.microsoft.com/dotnet/sdk:8.0 AS build
ARG configuration=Release
WORKDIR /src
COPY ["src/JT808.Protocol/JT808.Protocol.csproj", "src/JT808.Protocol/"]
RUN dotnet restore "src/JT808.Protocol/JT808.Protocol.csproj"
COPY . .
WORKDIR "/src/src/JT808.Protocol"
RUN dotnet build "JT808.Protocol.csproj" -c $configuration -o /app/build

FROM build AS publish
ARG configuration=Release
RUN dotnet publish "JT808.Protocol.csproj" -c $configuration -o /app/publish /p:UseAppHost=false

FROM base AS final
WORKDIR /app
COPY --from=publish /app/publish .
ENTRYPOINT ["dotnet", "JT808.Protocol.dll"]

```
</p>
</details> 

<details><summary>generated docker-compose.yml</summary>
<p>

```
version: '3.4'

services:
  jt808protocol:
    image: jt808protocol
    build:
      context: .
      dockerfile: src/JT808.Protocol/Dockerfile
```

</p>
</details>

yedajiang44, 2023-12-07:
This project is a library, not an application, so it can't be run directly, but I'd like to know why you need to run it @rockenbach

yedajiang44, 2023-12-07:
I forgot to tell you that you can debug or run unit tests in vscode, they are in `src/JT808.Protocol.Test`, but you need to install some extensions, otherwise you may not be able to view the test output
@rockenbach