SmallChi/JT808 #35: 答应数据 分包的问题

leida1983, 2024-02-21:
/// <summary>
    ///     请求注册的二维码
    /// </summary>
    /// <param name="request"></param>
    /// <returns></returns>
    public byte[] Msg0xE003(JT808HeaderPackage request, IJT808Session session)
    {
        if (request.Bodies == null)
        {
            logger.LogError("数据体为空");
            return default;
        }

      //   下面的 ImageData  是 一个 100多K 的 图像的数据  我自定义的 MsgId

        var splitPackage = JT808Config.SplitPackageStrategy.Processor(ImageData).ToList();
        ushort packgeCount = (ushort)splitPackage.Count();
        List<string> hexs = new List<string>();
        for (ushort i = 1; i <= packgeCount; i++)
        {
            JT808Package package = new JT808Package
            {
                Header = new JT808Header
                {
                    MsgId = 0xF003,
                    ManualMsgNum = request.Header.MsgNum,
                    TerminalPhoneNo = request.Header.TerminalPhoneNo,
                    MessageBodyProperty = new JT808HeaderMessageBodyProperty()
                    {
                        IsPackage = true,
                    },
                    PackgeCount = packgeCount,
                    PackageIndex = i
                },
                SubDataBodies = splitPackage[i - 1].Data
            };
            var hex = JT808Serializer.Serialize(package, request.Version).ToHexString();
            hexs.Add(hex);
        }

        foreach (var s in hexs)
        {
            session.SendAsync(s.ToHexBytes());
//   如果  我这里不加上线程的休息   会导致 客户端 收不到全部的数据
            global::System.Threading.Thread.Sleep(200);
        }

        return default;
    }

我觉得这个 Sleep  不是很合理  ，有没有 更好的方法呢！！

SmallChi, 2024-02-23:
这个跟编码没有关系吧，是不是设备端处理的方式有问题呢，没有处理连包问题？

leida1983, 2024-02-24:
不知道是不是太大了  100k的   分成100多个包  中间偶发 包会解析失败 

回到前面问题   调用的时候返回多包  您是怎么写的啊  愿意付费 咨询   但是你们qq没人

yedajiang44, 2024-02-24:
- > 如果 我这里不加上线程的休息 会导致 客户端 收不到全部的数据
  
  这是客户端问题，与你无关

- > 我觉得这个 Sleep 不是很合理 ，有没有 更好的方法呢！！
  
  移除相关代码，让客户端自己排查

- > 不知道是不是太大了 100k的 分成100多个包 中间偶发 包会解析失败
  
  与数据大小无关，与分包数量无关，同上让客户端自己排查

- > 调用的时候返回多包 您是怎么写的
  
  这个需要根据实际情况而定，没有固定的写法，也没有所谓的模板，因为这与业务挂钩

大致看了下你的代码，可能你对协议文档或者808机制不熟，稍微调整了下，伪代码如下
```c#
        public List<byte[]> AutoSplitPackage(string simNumber, byte[] data, IJT808Config config)
        {
            var splitPackage = config.SplitPackageStrategy.Processor(data).ToList();
            var result = new List<byte[]>();
            for (ushort i = 1, packgeCount = (ushort)splitPackage.Count; i <= packgeCount; i++)
            {
                var package = new JT808Package
                {
                    Header = new JT808Header
                    {
                        MsgId = 0xF003,
                        TerminalPhoneNo = simNumber,
                        MessageBodyProperty = new JT808HeaderMessageBodyProperty()
                        {
                            IsPackage = true,
                        },
                        PackgeCount = packgeCount,
                        PackageIndex = i
                    },
                    SubDataBodies = splitPackage[i - 1].Data
                };
                result.Add(config.GetSerializer().Serialize(package));
            }
            return result;
        }
```

closed 2024-03-21 by SmallChi