SmallChi/JT808 #38: messageid not found

ankushgowda2026, 2024-08-08:
serializer.Serialize(JT808MsgId._0x1205.CreateCustomMsgId2019(Client?.DeviceCode, new JT808_0x1205()

'JT808MsgId' does not contain a definition for '_0x1205'

yedajiang44, 2024-08-08:
给一个复现的单元测试

yedajiang44, 2024-08-08:
Can you give a reproducible unit test? @ankushgowda2026

ankushgowda2026, 2024-08-08:
request = serializer.Serialize((JT808MsgId._0x1205).CreateCustomMsgId2019(Client?.DeviceCode, new JT808_0x1205()
                    {
                        AlarmFlag=1,
                        AVResourceType=2,
                        BeginTime=Convert.ToDateTime("2024-08-08 10:20:01"),
                        EndTime=Convert.ToDateTime("2024-08-08 10:50:01"),
                        FileSize=3,
                        LogicChannelNo=(byte)channel,
                        MemoryType=5,
                        StreamType=6
                        
                    }));
                    
                    Hi @yedajiang44 i am trying to upload the video from the perticular device

ankushgowda2026, 2024-08-08:
Thank you, @yedajiang44 I am able to resolve the problem using packets, but I have a small doubt. Where do I need to point (ftpserver or cms server)? If I point to any server, I am not passing any port or IP, so how can I reseave the files?

yedajiang44, 2024-08-09:
This is a query, when the server sends 0x9205, you just need to return 0x1205. When the server sends 0x9206, then send the relevant resources to ftp.

closed 2024-08-12 by yedajiang44