SmallChi/JT808 #42: 好像没实现消息加密和解密？

china-live, 2024-11-15:
好像没实现消息加密和解密

SmallChi, 2024-11-15:
有用到加密和解码的？我没有遇到过了，有对应数据没有？方便提供看下吗？

china-live, 2024-11-15:
没有数据，就是现在行业监管越来越严了么，有将加密传输变成行业标准趋势。
看了一下，808确实也有加密传输的定义，只是好像目前大家都没实现这个，但后面怕是避免不了了。

china-live, 2024-11-15:
JTT808中对 0x8A00 和 0x0A00这两个指令只简单一句话，也不知道具体是怎么应用的。
初步猜测是：平台发送 0x8A00 将平台公钥发送给终端，表示启用加密传输功能，终端收到后，返回 0x0A00 将终端自己的公钥返回给平台。现在有两个不确定的是，1. 终端是收到  0x8A00 指令后，立即启用加密，还是在返回 0x0A00 再后面的数据发送再启用。 2。只要启用了加密传输，好像就没法解除了，最多只能更换新的公钥。

SmallChi, 2024-11-15:
不太清楚了，等具体要求下来才知道了，走一步算一步。

closed 2024-11-15 by SmallChi

china-live, 2024-11-15:
还有一个麻烦的是，这样平台打包和解包必须和会话管理强关联了。

SmallChi, 2024-11-15:
不用想那么多，走到那一步在说了，似在非在，没有搞部标。

china-live, 2024-11-15:
已经在调研了，说不定那天就来了。这TM一来，现在的系统影响和改动还是相当大的，早准备一下。

SmallChi, 2024-11-15:
遇事不要慌，先喝喝茶，压压惊，在钓钓鱼，等出来再说。不至于一下子一刀切。

SmallChi, 2024-11-19:
@wosledon 在深圳吗？

wosledon, 2024-11-19:
> @wosledon 在深圳吗？

在的在的

china-live, 2024-11-19:
不一样，国电那个GB28181和国网B接口和这个没直接关系。这个要真的实施起来的话，影响还是蛮大的。

SmallChi, 2024-11-19:
@wosledon 加你qq了，通过以下。