SmallChi/JT808 #44: configure the parameters doubt

ankushgowda2026, 2024-12-09:
public void setpara()
        {
            var package = new JT808Package
            {
                Header = new JT808Header
                {
                    MsgId = 0x8103,
                    TerminalPhoneNo = "251190000500", 
                    MsgNum = 1 
                },
                Bodies = new JT808_0x8103
                {
                    ParamList = new List<JT808_0x8103_BodyBase>
                    {
                        new JT808_0x8103_0x0001
                        {
                            ParamValue = 20 
                        }
                    }
                }
            };
            serializer.Serialize(package);
        }
       Hello, I'm attempting to configure the terminal heartbeat parameters to update every 20 seconds, but it's not updating. Perhaps the structure I'm sending is incorrect, if so, could someone please help? I also want to know how to get any parameter with value.

waterliu99, 2024-12-10:
There is no problem with the demo you provided. You need to check whether the device supports setting the heartbeat 20s. What is the logic that has triggered the heartbeat.  you can print this  hex string let me see.

ankushgowda2026, 2024-12-10:
public void setpara(string deviceCode )
        {
            var package = new JT808Package
            {
                Header = new JT808Header
                {
                    MsgId = 0x8103,
                    TerminalPhoneNo = deviceCode, 
                    MsgNum = 1 
                },
                Bodies = new JT808_0x8103
                {
                    ParamList = new List<JT808_0x8103_BodyBase>
                    {
                        new JT808_0x8103_0x0001
                        {
                            ParamValue = 20 
                        }
                    }
                }
            };
            var hex = JT808Serializer.Serialize(package).ToHexString();

            Console.WriteLine(hex);
        }

hi this is the hex i am getting 
7E8103000A000000000000000101000000010400000014997E

waterliu99, 2024-12-10:
this hex string is correct ,you need to confirm your device supporting set heartbeat

waterliu99, 2024-12-10:
you need to ask your device supplier

ankushgowda2026, 2024-12-10:
Ok I want set adas and dms events settings how can I do that, I am getting those events in 65 and 64 of 200 message body.

waterliu99, 2024-12-10:
1.

waterliu99, 2024-12-10:
1. install JT808.Protocol.Extensions.SuBiao
2. register SuBiao service
   2.1  DI register::
          IServiceCollection serviceDescriptors1 = new ServiceCollection();
          serviceDescriptors1.AddJT808Configure()
                                        .AddSuBiaoConfigure();

  2.2 global register:
       JT808Serializer.Instance.Register(JT808_SuBiao_Constants.GetCurrentAssembly());

ankushgowda2026, 2024-12-10:
i have done this part and i am getting events but i need to change the configuration
for example(distraction speed, interval...)

SmallChi, 2024-12-12:
Consult your equipment supplier first to see if the command settings for these parameters are supported.

ankushgowda2026, 2024-12-12:
ok, thank you @waterliu99 and @SmallChi for support

closed 2024-12-12 by ankushgowda2026