SmallChi/JT808 #46: sending Acknowledge to AlarmFlag

ankushgowda2026, 2025-02-25:
Hi Team, I am stuck on an issue where I continuously receive the event from JT808_0x0200.AlarmFlag (Emergency Alarm) at intervals. I tried acknowledging it with ID _0x8001, but it did not work. How should I properly acknowledge this AlarmFlag.

yedajiang44, 2025-02-25:
take your 0x8001 hex here

ankushgowda2026, 2025-02-27:
sorry for the delay response

if ((JT808Version)Client.jT808Version == JT808Version.JTT2019)
                    {
                        response = serializer.Serialize(JT808MsgId._0x8001.Create2019(Client?.DeviceCode, new JT808_0x8001
                        {
                            AckMsgId = request.Header.MsgId,
                            JT808PlatformResult = JT808PlatformResult.succeed,
                            MsgNum = request.Header.MsgNum
                        }));
                    }
                    else
                    {
                        response = serializer.Serialize(JT808MsgId._0x8001.Create(Client?.DeviceCode, new JT808_0x8001
                        {
                            AckMsgId = request.Header.MsgId,
                            JT808PlatformResult = JT808PlatformResult.succeed,
                            MsgNum = request.Header.MsgNum
                            // MsgNum = request.Header.MsgNum
                        }));
                    }

hex 7E80014005010000000025119000050000150015020000667E

yedajiang44, 2025-02-27:
You should reply 0x8203

yedajiang44, 2025-03-01:
You need to read the jt808 documentation in detail so that you can understand the meaning of each attribute.
 
- ManualConfirmAlarmType is the value of alarm bitwise operation parsings of 0x0200
- AlarmMsgNum must be from JT808Package containing 0x0200
- AlarmMsgNum = 0, does not affect other alarm messages

ankushgowda2026, 2025-03-01:
Thank you, @yedajiang44

closed 2025-03-01 by ankushgowda2026