SmallChi/JT808 #47: Not Received Video Stream Packet

h128, 2025-03-01:
Hello, 
I recently bought some vehicle device that use JT808, I have issue regarding not receiving Video Packet data. in the following I will share my code. maybe you can help me. also it would be really helpful if you share some sample code with me regarding that.

`
using JT808.Gateway;
using JT808.Gateway.Abstractions;
using JT808.Gateway.Transmit;
using JT808.Protocol;
using JT808.Protocol.Enums;
using JT808.Protocol.Extensions;
using JT808.Protocol.Extensions.JT1078;
using JT808.Protocol.Extensions.JT1078.MessageBody;
using JT808.Protocol.MessageBody;
using SampleDash;

var builder = Host.CreateApplicationBuilder(args);
builder.Services
    .AddJT808Configure()
    .AddJT1078Configure()
    .AddGateway(builder.Configuration)
    .AddMessageHandler<JT808MessageHandlerImpl>()
    .AddTransmit(builder.Configuration)
    .AddTcp()
    .AddUdp()                ;

builder.Services.AddHostedService<Worker>();

var host = builder.Build();

host.Run();

public class JT808MessageHandlerImpl : JT808MessageHandler
{
    private readonly ILogger logger;
    private readonly JT808TransmitService jT808TransmitService;

    public JT808MessageHandlerImpl(
        ILoggerFactory loggerFactory,
        JT808TransmitService jT808TransmitService,
        IJT808Config jT808Config)
        : base(jT808Config)
    {
        this.jT808TransmitService = jT808TransmitService;
        logger = loggerFactory.CreateLogger<JT808MessageHandlerImpl>();

    }

    public byte[] SendVideoStartCommand(string terminalPhoneNo, byte channelNo)
    {
        var command = new JT808_0x9101
        {
            ServerIp = "79.132.193.50", // Change this to your actual IP
            UdpPort = 1078,  // Ensure UDP port is open
            TcpPort = 0,
            ChannelNo = channelNo,
             
            DataType = 0,       // 0: Main stream, 1: Sub stream
            StreamType = 0      // 0: Real-time, 1: Replay, 2: Download
        };

        var package = new JT808Package
        {
            Header = new JT808Header
            {
                MsgId = 0x9101,
                TerminalPhoneNo = terminalPhoneNo,
                  
            },
            Bodies = command
        };

        return JT808Serializer.Serialize(package, JT808Version.JTT2013);
    }

  

    public override byte[] Processor(in JT808HeaderPackage request)
    {
        logger.LogInformation($"Received JT808 Message: ID {request.Header.MsgId:X4}, Terminal: {request.Header.TerminalPhoneNo}");
        if (request.Header.MsgId == 0x0102)
        {
            return  JT808Serializer.Serialize(JT808MsgId._0x8001.Create(request.Header.TerminalPhoneNo, new JT808_0x8001
            {
                AckMsgId = request.Header.MsgId,
                JT808PlatformResult = JT808PlatformResult.succeed,
                MsgNum = request.Header.MsgNum
            }));

        }
        else if (request.Header.MsgId ==  0x0004)
        {
            return JT808Serializer.Serialize(JT808MsgId._0x8004.Create(request.Header.TerminalPhoneNo, new JT808_0x8004
            {
                Time = DateTime.Now,
            }));
        }
        else if (request.Header.MsgId == 0x0002)
        {
            var bytes2 = SendVideoStartCommand(request.Header.TerminalPhoneNo, channelNo: 1);
            jT808TransmitService.SendAsync((request.Header.TerminalPhoneNo, bytes2));

            var bytes = SendVideoStartCommand(request.Header.TerminalPhoneNo, channelNo: 2);
            jT808TransmitService.SendAsync((request.Header.TerminalPhoneNo, bytes));

            

            return JT808Serializer.Serialize(JT808MsgId._0x8001.Create(request.Header.TerminalPhoneNo, new JT808_0x8001
            {
                AckMsgId = request.Header.MsgId,
                JT808PlatformResult = JT808PlatformResult.succeed
            }));
        }
       
        switch (request.Header.MsgId)
            {
                // If using JT1078 directly
                case 0x1078: // Video packet for JT1078
                    Console.WriteLine("Video packet received");
                    // Process video data
                    break;
                case 0x9101: // Start Video Stream
                    Console.WriteLine("Start Video Stream");
                    break;
                case 0x9102: // Stop Video Stream
                    Console.WriteLine("Stop Video Stream");
                    break;
                case 0x9105: // Control Video Stream
                    Console.WriteLine("Control Video Stream");
                    break;
                case 0x9201: // Query Video Resources
                    Console.WriteLine("Query Video Resources");
                    break;
                case 0x9202: // Upload Stored Video
                    Console.WriteLine("Upload Stored Video");
                    break;
            }
        return base.Processor(request);
    }

}`

Log
`info: JT808.Gateway.JT808TcpServer[0]
      JT808 TCP Server start at 0.0.0.0:8808.
info: JT808.Gateway.Services.JT808TcpReceiveTimeoutHostedService[0]
      [Check Receive Timeout]
info: JT808.Gateway.Services.JT808TcpReceiveTimeoutHostedService[0]
      [Session Online Count]:0
info: JT808.Gateway.JT808UdpServer[0]
      JT808 Udp Server start at 0.0.0.0:1078.
info: JT808.Gateway.Services.JT808UdpReceiveTimeoutHostedService[0]
      [Check Receive Timeout]
info: JT808.Gateway.Services.JT808UdpReceiveTimeoutHostedService[0]
      [Session Online Count]:0
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Development
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Users\Administrator\source\repos\SampleDash\SampleDash
info: JT808.Gateway.JT808TcpServer[0]
      [Connected]:82.180.200.218:42293
info: JT808MessageHandlerImpl[0]
      Received JT808 Message: ID 0100, Terminal: 70109318407
info: JT808MessageHandlerImpl[0]
      Received JT808 Message: ID 0102, Terminal: 70109318407
info: JT808MessageHandlerImpl[0]
      Received JT808 Message: ID 0704, Terminal: 70109318407
info: JT808MessageHandlerImpl[0]
      Received JT808 Message: ID 0200, Terminal: 70109318407
info: JT808MessageHandlerImpl[0]
      Received JT808 Message: ID 0004, Terminal: 70109318407
info: JT808MessageHandlerImpl[0]
      Received JT808 Message: ID 0704, Terminal: 70109318407
info: JT808MessageHandlerImpl[0]
      Received JT808 Message: ID 0200, Terminal: 70109318407
info: JT808MessageHandlerImpl[0]
      Received JT808 Message: ID 0002, Terminal: 70109318407
info: JT808MessageHandlerImpl[0]
      Received JT808 Message: ID 0200, Terminal: 70109318407`

SmallChi, 2025-03-07:
Is the order issued incorrect? Take a look at the network card to capture the packet?

h128, 2025-03-07:
my question is the place SendVideoStartCommand after receiving 0x0002 is correct. and packet of video come to Processor function or not? I will install WireShark and inform you about the result after that.

SmallChi, 2025-03-09:
Problem solved?

closed 2025-03-15 by h128