SmallChi/JT808 #48: creating coustom body

ankushgowda2026, 2025-03-17:
Hi team,

I am currently working on implementing the JT808_0x0200_0x70 class to handle the AggressiveDrivingAlarm from UnknownLocationAttachData. I’ve received the following data payload:

UnknownLocationAttachData ID: 112
Value: 70-2F-00-00-00-10-00-02-00-05-00-5A-00-00-31-00-00-00-C5-4D-34-04-9F-C2-B5-25-03-11-17-06-33-04-01-00-00-00-00-00-00-00-25-03-11-17-06-33-00-06-00

I need to decode this data in the Analyze() function to extract GPS information, event images, and video data.

I tried comparing this payload with 0x65 data:
7B-22-41-74-74-61-63-68-49-6E-66-6F-49-64-22-3A-31-30-31-2C-22-41-74-74-61-63-68-49-6E-66-6F-4C-65-6E-67-74-68-22-3A-34-37-7D

however i don't understand much except datetime 25-03-11-17-06-33. I’d appreciate guidance on how to properly decode this message body.

SmallChi, 2025-03-17:
1.  Is it possible to provide the original 808 original hexadecimal data；
2. Check with the device provider whether the additional data 0x0200 is a private or active security protocol；
3. You can try using [JTTools](https://jttools.smallchi.cn/) to parse the appendages of various 808 extensions to see if they can match.

ankushgowda2026, 2025-03-17:
Hi thank you for your help, and give me some time so i get the  raw hexadecimal data from 808. and is there any stimulator for 0x70

Dozorov, 2025-03-17:
@ankushgowda2026 try:

Create file at `CustomMessageBody/JT808_0x0200_0x70.cs`

```csharp
using JT808.Protocol.Extensions.SuBiao.Metadata;
using JT808.Protocol.Formatters;
using JT808.Protocol.Interfaces;
using JT808.Protocol.MessageBody;
using JT808.Protocol.MessagePack;
using System;
using System.Text.Json;

namespace JT808.Protocol.Extensions.FELD_Custom
{
    public class JT808_0x0200_0x70 : JT808MessagePackFormatter<JT808_0x0200_0x70>, JT808_0x0200_CustomBodyBase, IJT808Analyze, IJT808_2019_Version
    {
        public byte AttachInfoId { get; set; } = 0x70;
        public byte AttachInfoLength { get; set; } = 31;
        public uint AlarmId { get; set; }
        public byte FlagState { get; set; }
        public byte AlarmOrEventType { get; set; }
        public ushort AlarmTimeThreshold { get; set; }
        public ushort AlarmThreshold1 { get; set; }
        public ushort AlarmThreshold2 { get; set; }
        public byte Speed { get; set; }
        public ushort Altitude { get; set; }
        public int Latitude { get; set; }
        public int Longitude { get; set; }
        public DateTime AlarmTime { get; set; }
        public ushort VehicleState { get; set; }
        public AlarmIdentificationProperty AlarmIdentification { get; set; }

        public void Analyze(ref JT808MessagePackReader reader, Utf8JsonWriter writer, IJT808Config config)
        {
            JT808_0x0200_0x70 value = new JT808_0x0200_0x70();
            value.AttachInfoId = reader.ReadByte();
            writer.WriteNumber("附加信息Id", value.AttachInfoId);
            value.AttachInfoLength = reader.ReadByte();
            writer.WriteNumber("附加信息长度", value.AttachInfoLength);
            value.AlarmId = reader.ReadUInt32();
            writer.WriteNumber("报警ID", value.AlarmId);
            value.FlagState = reader.ReadByte();
            writer.WriteNumber("标志状态", value.FlagState);
            value.AlarmOrEventType = reader.ReadByte();
            writer.WriteNumber("报警/事件类型", value.AlarmOrEventType);
            value.AlarmTimeThreshold = reader.ReadUInt16();
            writer.WriteNumber("报警时间阈值(秒)", value.AlarmTimeThreshold);
            value.AlarmThreshold1 = reader.ReadUInt16();
            writer.WriteNumber("报警阈值1", value.AlarmThreshold1);
            value.AlarmThreshold2 = reader.ReadUInt16();
            writer.WriteNumber("报警阈值2", value.AlarmThreshold2);
            value.Speed = reader.ReadByte();
            writer.WriteNumber("车速(Km/h)", value.Speed);
            value.Altitude = reader.ReadUInt16();
            writer.WriteNumber("高度(m)", value.Altitude);
            value.Latitude = (int)reader.ReadUInt32();
            writer.WriteNumber("纬度", value.Latitude);
            value.Longitude = (int)reader.ReadUInt32();
            writer.WriteNumber("经度", value.Longitude);
            value.AlarmTime = reader.ReadDateTime_yyMMddHHmmss();
            writer.WriteString("报警时间", value.AlarmTime.ToString("yyyy-MM-dd HH:mm:ss"));
            value.VehicleState = reader.ReadUInt16();
            writer.WriteNumber("车辆状态", value.VehicleState);
            value.AlarmIdentification = new AlarmIdentificationProperty();
            string terminalIDHex = reader.ReadVirtualArray(7).ToArray().ToHexString();
            value.AlarmIdentification.TerminalID = reader.ReadString(7);
            value.AlarmIdentification.Time = reader.ReadDateTime_yyMMddHHmmss();
            value.AlarmIdentification.SN = reader.ReadByte();
            value.AlarmIdentification.AttachCount = reader.ReadByte();
[... 60 lines cut ...]
            writer.WriteByte(value.AlarmIdentification.AttachCount);
            writer.WriteByte(value.AlarmIdentification.Retain);
            writer.WriteByteReturn((byte)(writer.GetCurrentPosition() - AttachInfoLengthPosition - 1), AttachInfoLengthPosition);
        }
    }
}

And configuration:

public class FELDConfigBase : GlobalConfigBase
{
    public override string ConfigId { get; protected set; } = "FELDConfig";
}

IJT808Config JT808Config = new FELDConfigBase();
JT808Config.JT808_0X0200_Custom_Factory.SetMap<JT808_0x0200_0x70>();
JT808Config.FormatterFactory.SetMap<JT808_0x0200_0x70>();

JT808Serializer = JT808Config.GetSerializer();

And use JT808Serializer to Serilize and Deserilize you data.

@ankushgowda2026 Do you have an Email or Telegram? Im workinf on the same tasks and we can help each other.

My Telegram: https://t.me/macmillanusa

ankushgowda2026, 2025-03-18:
thank you @Dozorov i will connect with you on telegram

closed 2025-03-18 by ankushgowda2026