SmallChi/JT808 #49: Command Relay JT808

lucasmartinshilario, 2025-03-18:
Hello guys, all right?

I’m having problems sending the relay action command (relay,1# to activate and relay,0# to deactivate) via ip connection.
Do not know which Message ID and Message Body to put, this in HEX format. 
Could someone help me with this?

SmallChi, 2025-03-19:
Please describe your problem in detail.
1.What message ID to use;
2.Provide complete HEX data.

lucasmartinshilario, 2025-03-19:
@SmallChi Thank you for the brief feedback, but after several tests I was able to find the right command on the server I was using.
I leave here registered if someone needs the commands Relay,1 and Relay,0.

Message ID: 8500
Message Body: 01 (Relay,1#) or 00 (Relay,0#)

Thank you!

closed 2025-03-19 by lucasmartinshilario